alexandrebarachant/muse-lsl #136: Using a newer version of pylsl

ErikBjare, 2020-10-11:
I was trying some examples in the pylsl repo but a lot of stuff broke on me since it relies on newer features in pylsl.

I found the version lock to 1.10.1 on Linux systems, and after checking git blame I saw it was introduced in https://github.com/alexandrebarachant/muse-lsl/pull/77.

Does the original issue still persist on recent releases of pylsl? Would the maintainers be willing to accept a change fixing this if I get it working?

ErikBjare, 2020-10-26:
@JohnGriffiths is this of interest to you? 

I notice that when I manually install a newer version some stuff breaks in OpenBCI device detection code at import-time in https://github.com/NeuroTechX/eeg-notebooks (very surprising), but otherwise things seem to work fine.

ErikBjare, 2020-11-12:
As I replied in https://github.com/alexandrebarachant/muse-lsl/issues/72#issuecomment-726112459, issues still persist with newer versions of `pylsl`.

xloem, 2022-03-06:
It sounds like this may be an easy to fix issue. It would be great to narrow down how a developer can reproduce it.

xloem, 2022-03-10:
I propose pylsl be upgraded. The underlying issue was a manylinux + boost packaging issue producing system conflicts in the embedded .so . pylsl no longer embeds a .so in the linux wheels, rather documenting that the user should install liblsl separately, thus preventing system conflicts.  Hence, this issue should no longer occur, and if a similar one does it is likely a problem with the user's installation of liblsl.

jdpigeon, 2022-03-11:
Having to pin lsl was an unfortunate side effect of the longstanding .so issue that @xloem describes. If it is truly fixed and a new pylsl is released I would be more than happy to upgrade it

xloem, 2022-12-11:
I believe the liblsl dependency should be upgraded. There have been many new releases since the failing one, and the problem was upstream with liblsl, not with muse-lsl.

closed 2026-09-08 by jdpigeon