alexandrebarachant/muse-lsl #137: muselsl list issue on Raspberry Pi

priyanshin, 2020-10-14:
Hi! We are part of Neurotech @ Davis working on a project to try to connect and stream data from the muse2 to the raspberry pi and are encountering the following error.

pi@raspberrypi:~ $ muselsl list
Traceback (most recent call last):
  File "/home/pi/.local/bin/muselsl", line 5, in <module>
    from muselsl.__main__ import main
  File "/home/pi/.local/lib/python3.7/site-packages/muselsl/__init__.py", line 1, in <module>
    from .stream import stream, list_muses
  File "/home/pi/.local/lib/python3.7/site-packages/muselsl/stream.py", line 2, in <module>
    from pylsl import StreamInfo, StreamOutlet
  File "/home/pi/.local/lib/python3.7/site-packages/pylsl/__init__.py", line 2, in <module>
    from .pylsl import IRREGULAR_RATE, DEDUCED_TIMESTAMP, FOREVER, cf_float32,\
  File "/home/pi/.local/lib/python3.7/site-packages/pylsl/pylsl.py", line 1147, in <module>
    lib = CDLL(libpath)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /home/pi/.local/lib/python3.7/site-packages/pylsl/liblsl32.so: cannot open shared object file: No such file or directory

----------
We also noticed that this isn't specific to the muselsl list command. Any command we type using muselsl results in the same error. 

versions:
RaspberryPi = 10.4

Please let us know if you have encountered anything like this!

Best,
Priyanshi

oori, 2020-10-26:
Replace `liblsl32.so` in `pylsl` (from `liblsl-bcm2708.so`)
  1. `cd /home/pi/.local/lib/python3.7/site-packages`
  2. `wget ftp://sccn.ucsd.edu/pub/software/LSL/SDK/liblsl-C-C++-1.11.zip`
  3. `unzip liblsl-C-C++-1.11.zip`
  4. `sudo cp liblsl-bcm2708.so /home/pi/.local/lib/python3.7/site-packages/pylsl/liblsl32.so`

Ronit-Desai, 2020-11-11:
Hello, did you find a solution to this? I've been getting an error in the module lillsl64.dll module not found or try using the entire path in constructor syntax. While the file is already there at the mentioned path. Can you help me?

plnnr, 2021-02-15:
Following some of the advice from [this page](http://www.flaviofrohlich.org/thoughts/collecting-eeg-signals-with-a-raspberry-pi-and-a-muse-2-headset) but modifying the code:

1. Download the liblsl repository into any directory.
`git clone https://github.com/sccn/liblsl.git`
2. Enter the directory and begin the build (must have cmake installed via `sudo apt-get install cmake` or something like that):
```
cd liblsl
mkdir build
cd build
```
3. The issue I had with the tutorial was that it used `make` instead of `cmake`, and also I had to build it after `cmake`ing it.
```
cmake .. -DCMAKE_CXX_COMPILER=arm-linux-gnueabihf-g++ -DCMAKE_SHARED_LINKER_FLAGS="-Wl,--allow-multiple-definition"
cmake --build .
## Notice the period after build
sudo cp liblsl.so.1.14.0 /home/USER_NAME/.local/lib/python3.7/site-packages/pylsl/liblsl32.so
```
4. Then my scipy package needed fixing, and the advice was:
`sudo apt-get install libatlas-base-dev`

I can now run things in the command line like `muselsl stream` and then in a new command line `muselsl record`. Unfortunately functions like `view` don't work.

@Ronit-Desai I think your question is about a Windows machine though, based on your question containing something about a .dll file. This thread is specific to Rapsberry Pi 4, presumably running linux.

PeterGamma, 2021-06-18:
**There are still issues with the Rasperri PI 4:**

«I have also tried to stream Muse2 signal on RPi4 8GB but my stream also Disconnects. The problem may be in the communication via Bluetooth on Pi. Consider using dongle.»

**The stream is confirmed working with the Raspberry Pi 3B:**

«The stream works great with no Disconnection on Raspberry Pi 3B.»

http://www.flaviofrohlich.org/thoughts/collecting-eeg-signals-with-a-raspberry-pi-and-a-muse-2-headset

Although not all functions seem to work on the Rasperri PI, I am happy about every little piece of information regarding LSL and Rasperri PI.