alexandrebarachant/muse-lsl #140: Issue with EEG Stream on Raspberry PI 4

flaviofrohlich, 2020-12-01:
Please be gentle - this is my first post on github asking for help :-)

After several days of struggling, I have figured out thanks to the amazing help provided here how to get muselsl to (almost!) work on a Raspberry PI 4. I have tried to write it up in a way that non experts can follow and replicate. I hope [this](http://www.flaviofrohlich.org/thoughts/collecting-eeg-signals-with-a-raspberry-pi-and-a-muse-2-headset) helps.

With this, I can connect and start a stream. However, the stream disconnects and I am unable to collect any data (details below).

I have extensively searched here and elsewhere and have not found anything that helped. I am hoping a kind soul has some ideas how to proceed from here. Thank you!!!

Most simple version of the script:

muses = list_muses()
stream(muses[0]['address'], ppg_enabled=False, acc_enabled=False, gyro_enabled=False)

Searching for Muses, this may take up to 10 seconds...
Found device Muse-****, MAC Address **:**:**:**:**:**
2020-12-01 17:57:38.097 (  13.820s) [        B6EFEAD0]             common.cpp:50    INFO| git:/branch:/build:/compiler:GNU-8.3.0/link:shared
Connecting to Muse: **:**:**:**:**:**...
Connected.
Streaming EEG...
Disconnected.

oori, 2020-12-07:
@flaviofrohlich I emailed you back.    Start with reading this issue https://github.com/alexandrebarachant/muse-lsl/issues/110#issuecomment-566017757
It's mostly about using BLED112, BGAPI and avoiding PPG+AAC when not needed.

Good luck.

meeththakkar, 2021-01-07:
Small world Dr. Frohlich :)

Subscribing.

flaviofrohlich, 2021-01-08:
Wow - indeed. Good to hear from you (please send me an email to reconnect!). Just as a quick update. I would like to thank everyone who has reached out. I have tried numerous things, including using the BLED dongle - yet still the stream does not contain any data and immediately disconnects. I have few more leads to pursue but will keep posting here in case there is progress. I am assuming there is a small but important thing that i am missing. I am quite rusty in anything coding but enjoy learning from you. Thank you all!

JohnGriffiths, 2021-02-13:
New raspberry pi arrived. Installed muselsl, replicated this error.

JohnGriffiths, 2021-02-13:
Working fix appears to be, with BLED112 dongle installed:

`muselsl stream -b bgapi`

JohnGriffiths, 2021-02-16:
FYI, I'm collecting installation notes and other misc code and info for raspberry pi `muselsl` and `eeg-notebooks` in this (probably temporay) repo

https://github.com/JohnGriffiths/eegnb_rpi/

See `installation_notes.md` for an apparently functioning installation procedure.

PeterGamma, 2021-06-12:
@JohnGriffiths: thanks for your support. I hope the Muse 2 will be soon a well established research grade device without issues.

PeterGamma, 2021-06-18:
**There are still issues with the Rasperri PI 4:**

«I have also tried to stream Muse2 signal on RPi4 8GB but my stream also Disconnects. The problem may be in the communication via Bluetooth on Pi. Consider using dongle.»

**The stream is working with the  Raspberry Pi 3B**:

«The stream works great with no Disconnection on Raspberry Pi 3B.»

http://www.flaviofrohlich.org/thoughts/collecting-eeg-signals-with-a-raspberry-pi-and-a-muse-2-headset

buyuk-dev, 2021-08-15:
I've had similar issue while running my modified muselsl on macOS. Turned out the timeout in muselsl/stream.py while loop after starting the stream was too small (perhaps low bitrate with our BLE modules). Try increasing AUTO_DISCONNECT_DELAY in muselsl/constants.py. Or replace the condition in while loop with infinite loop (while True:)

ChrisSwan, 2021-12-06:
Has anyone found a solution to this issue? I tried getting my Pi 4 set up over the weekend and have the same problem of the muse stream immediately disconnecting.

I've tried incresaing the AUTO_DISCONNET_DELAY, and the changing to an infinite loop, but neither approach seemed to work.

ChrisSwan, 2021-12-12:
Just adding some more info, although I haven't found the solution yet. This is with  Pi4, using onboard bluetooth, and trying the startStream example (so just EEG):

* Watching the Muse light I can see it goes solid as the connection happens, but then goes back into pairing mode about a second later.
* Adding some prints in various places I can see that _handle_eeg() does start to receive data ok, but even with an infinite loop a disconnect still occurs
* The actual disconnect() method is not being called
* I tried adding a keep_alive() call in the infinite loop (before the sleep), and watching the write_cmd() I can see that 'k' is correctly sent twice, but I then crash with a pygatt.exceptions.NotConnectedError

Hannahke123, 2022-10-27:
Hi, Is anyone getting an error like this?

Connecting to Muse: 00:55:DA:B9:67:2E...
Connected.
Streaming EEG...
ERROR:root:A message handler raised an exception: unsupported operand type(s) for +: 'BleakGATTCharacteristicBlueZDBus' and 'int'.
Traceback (most recent call last):
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/dbus_fast/message_bus.py", line 778, in _process_message
    result = user_handler(msg)
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/bleak/backends/bluezdbus/manager.py", line 832, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 152, in on_value_changed
    callback(bytearray(value))
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/muselsl/backends.py", line 63, in wrap
    value_handle = declaration_handle + 1
TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicBlueZDBus' and 'int'

Do you have any solutions to this? Let me know!

PeterGamma, 2024-02-12:
Hi Flavio Frohlich,

I am interested in resolving the issue described here as well, altough I usually don’t code. As far as I have seen here, no one was yet able to resolve the issue discribed here. After studying devices like the Muse headband, I found the information that Lab Streaming Layer with Python code requires an x86 processor, butRasperry Pi’s have arm based processors.  There is code available which connect LSL to Python. But the code requires an x86 processor, which for instance a LattePanda has, but not Raspberry Pi’s

On a Raspberri Pi, LSL to Python does not run. This is an issue which has been described in the Lab Streaming Layer documentation of the Swartz Center of Computational Neuroscience. Unfortunately, I cannot refind where this has been written.  But there was an issue at the LabStreamingLayer GITHUB with Rasperrry Pi’s and LabStreamingLayer as well:

pylsl not working on RaspberryPi #36 

https://github.com/labstreaminglayer/pylsl/issues/36

Software developers at the Swartz Center of Computational Neuroscience https://sccn.ucsd.edu/  where the LabStreamingLayer has been developed use Matlab which requires an x86 processor. And I suppose they developed LSL with Matlab. But Matlab does not run on Raspberry Pi’s either. LSL was developed at https://sccn.ucsd.edu/, but also EEGLAB and BCILAB, and all of these software applications are based on Matlab.

To resolve the issue described here, I suggest for instance to try to use a LattePanda Delta instead of a Raspberry Pi 4, and to test if this works. 

If you want to say with the Rasperry Pi platform, I suggest to use instead of the Muse headband OpenBCI modules and BrainFlow:

https://medium.com/@basoph2002/using-brainflow-with-openbci-eeg-recorder-on-a-raspberry-pi-e578dca675ce

Or the PiEEG shield:

http://pieeg.com/

Which offers Python code which runs on the Raspberry Pi. PiEEG was first announced, but now it says that it is unavailable:

https://www.crowdsupply.com/hackerbci/pieeg#products

4 channels where available for a short time, but now it is gone. Please support me and write to the PiEEG developer Ildar Rakhmatulin, PhD. to bring it back. E-mail:

[I.Rakhmatulin@hw.ac.uk](https://researchportal.hw.ac.uk/en/persons/ildar-rakhmatulin#) 

But there are many more EEG, ECG, EMG, etc. devices available for developers who are on a budget. You can find those on the site:

[www.petergamma.org](http://www.petergamma.org/)

Peter Gamma is greeting  Flavio Frohlich from Switzerland, Peter Gamma from the Meditation Research Institute Switzerland (MRIS)

xloem, 2024-08-28:
My personal experience is that this is a problem with muselsl and the Muse at the Bluetooth level, or the local radio environment of the user, _not_ LSL nor Arm. When I ran into an issue like this it happened even without using LSL, such as direct recording to file.

PeterGamma, 2024-08-30:
Interaxon does not support 3.th party developments anymore. Interaxon has pulled the Muse SDK from the market, so it is not possible anymore the make your own development. Alexandre Barachants Muse Headband LabStreamingLayer interface is a reverse engineering development of Alexandre and his friend as he once said he made over the week-end. OpenBCI is more open regarding own developments. Here is an example how to connect OpenBCI Cyton to a  Raspberry Pi 3 model B+:

https://medium.com/@basoph2002/using-brainflow-with-openbci-eeg-recorder-on-a-raspberry-pi-e578dca675ce

The author writes it should also work with the Raspberry Pi 4. If you are looking for an affordable alternative to OpenBCI Cyton, you can try if you can fix OpenBCI modules which are sold on eBay and Aliexpress. You can find instructions how to this on my site [www.petergamma.org](http://www.petergamma.org/). 

An even better solution for the Raspberry Pi would be PiEEG, since it is a Raspberry Pi hat:

https://www.elecrow.com/pieeg.html?srsltid=AfmBOop79s5U9EeL7FiCPD7nyZj4hXLTCNZZQsxnScj-YApnvgDC_jUY

But PiEEG is new and there are rumors that it has issues with noise. Please help me if you can and test PiEEG, publish issues in the PiEEG forum if you find such:

https://pieeg.com/forum-pieeg-low-cost-brain-computer-interface/

and send PiEEG back to Elecrow if it has if it has issues. According my own little personal experience it is better to choose open platforms than a platform such as the Muse Headband which does not support 3 th party developments anymore.