alexandrebarachant/muse-lsl #147: Maintenance Status

xloem, 2021-01-07:
@alexandrebarachant I notice there's been no maintainer activity on this repository since June of last year.  Would you be willing to enable [automatic merging of pull requests](https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/automatically-merging-a-pull-request)?  It's configurable so you could require community reviews etc before merging if that's a concern.

xloem, 2021-01-07:
We've merged the outstanding PRs into https://github.com/xloem/muse-lsl/tree/dev , and resolved bugs locally, but that tree still needs others to test it.  What would help people have more capacity for pull request maintenance, merging, testing?

jdpigeon, 2021-01-07:
Hey @xloem, I'm very excited to see all the updates that you've made recently. Me and Alex would be happy to discuss how to handle maintenance of the library going forward, but in the mean time I will be happy to review and merge in some of your recent PRs this weekend. Our capacity for reviewing has been a little slow because most updates require us to actually test the library with real device, but we are still around and have a strong interest in supporting EEG research.