alexandrebarachant/muse-lsl #156: Issue in code when connecting to Muse 2

ary1234554321, 2021-02-20:
Hi, I tried running the given code in the GitHub explanation, 

from muselsl import stream, list_muses

muses = list_muses()
print(muses)
stream('The Mac address is here')

print('Stream has ended')

But when it runs, it gives this error,

Traceback (most recent call last):
  File "C:/Users/aryan/PycharmProjects/Muse/mtest.py", line 5, in <module>
    stream('00-55-da-b5-61-ff')
  File "C:\Users\aryan\PycharmProjects\Muse\venv\lib\site-packages\muselsl\stream.py", line 170, in stream
    while time() - muse.last_timestamp < AUTO_DISCONNECT_DELAY:
AttributeError: 'Muse' object has no attribute 'last_timestamp'

Does anyone have any idea why this is happening? I just want to get real-time data from my Muse 2.

jdpigeon, 2021-02-23:
Thanks for reporting this. I'm not sure why this is happening, but can look into it soon. In the meantime, can you connect to your Muse when using the CLI? e.g. `muselsl stream` from the terminal.

buyuk-dev, 2021-08-15:
This seems to be windows-only, so I looked at the muselsl.py/muse.py and in the Muse::connect() method for the BlueMuse the last_timestamp field was not set resulting in race resulting in this error. Here is my fix suggestions: https://github.com/alexandrebarachant/muse-lsl/pull/167