alexandrebarachant/muse-lsl #157: muselsl cannot find muse S head band

NathanTh0mas, 2021-04-14:
I installed muselsl and BlueMuse. BlueMuse has no issue detecting my Muse S, however, when I run the basic script below muselsl cannot find any headband. Please advise.

```
from muselsl import stream, list_muses

if __name__ == "__main__":

    muses = list_muses()

    if not muses:
        print('No Muses found')
    else:
        print(stream(muses[0]['address']))
```

The error states:
> "C:\Program Files\Python39\python.exe" D:/Muse/Python/main.py
> Starting BlueMuse, see BlueMuse window for interactive list of devices.
> No Muses found

rizaru22, 2022-02-07:
Hi @NathanTh0mas , 
Did you find the solution ? I face the same problem...

WeAreMidnight, 2023-10-20:
Same issue here!