alexandrebarachant/muse-lsl #160: MUSE data -> OpenBCI + Brainflow

sunshineinsandiego, 2021-05-11:
Hi - Thanks for this Muse package. I've been experimenting with it recently, and I'm wondering if it's possible to pipe data from MUSE to the OpenBCI GUI + Brainflow analysis software? I'm working with a few different headsets / biosensing devices and would like to simplify the analysis pipeline by using a common software stack, and it seems like the OpenBCI tools + Brainflow software is a good match. Currently, however, OpenBCI doesn't support the Muse headset. Do you have any recommendations on either (a) how to pipe data into the OpenBCI+Brainflow stack or (b) software stack that can handle a variety of different EEG + ECG sensors for analysis? Thanks!

jdpigeon, 2021-05-12:
I'm not too familiar with Brainflow, but it seems like it replaces and expands upon the capabilities of LSL, which this library is based on.

You may be able to pipe data into the OpenBCI GUI if it allows LSL input. However, I can't imagine it would work too well as it seems like it should be designed specifically for OpenBCI devices.

sunshineinsandiego, 2021-05-12:
Got it, thanks. Will try.

closed 2021-05-12 by sunshineinsandiego