alexandrebarachant/muse-lsl #176: Please add a specification on listening ip address

Zibri, 2022-03-21:
This does not work, probably because my pc has multiple ip addresses.
```
C:\WINDOWS\system32>python -m muselsl stream
Searching for Muses, this may take up to 10 seconds...
Found device Muse-E133, MAC Address 00:55:DA:B0:E1:33
2022-03-21 20:46:39.878 (  11.320s) [        53DA8980]         api_config.cpp:231   INFO| Loaded default config
2022-03-21 20:46:39.879 (  11.321s) [        53DA8980]             common.cpp:64    INFO| git:a0fc2fde92a10f4cb5fccbc5552228b865f17379/branch:refs/tags/v1.15.0/build:Release/compiler:MSVC-19.0.24245.0/link:SHARED
2022-03-21 20:46:39.883 (  11.325s) [        53DA8980] stream_outlet_impl.cpp:88    WARN| Couldn't create multicast responder for 224.0.0.1 (set_option: A socket operation was attempted to an unreachable host)
Connecting to Muse-E133: 00:55:DA:B0:E1:33...
Traceback (most recent call last):
  File "C:\Users\Zibri\AppData\Local\Programs\Python\Python310\lib\asyncio\locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Zibri\AppData\Local\Programs\Python\Python310\lib\asyncio\tasks.py", line 456, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

```
this instead works:
```
C:\>python -m muselsl record_direct
Searching for Muses, this may take up to 10 seconds...
Found device Muse-E133, MAC Address 00:55:DA:B0:E1:33
Connecting to Muse-E133 : 00:55:DA:B0:E1:33...
Connecting to Muse: 00:55:DA:B0:E1:33...
Start recording at time t=1647888538.976
Done - wrote file: C:\recording_2022-03-21-18.48.50.csv.

```

Zibri, 2022-03-22:
P.S.
The headband works perfectly.
The problem seems to be in python listening part.
record_direct works.

Zibri, 2022-03-23:
I have no idea what happened.
Now it works.

[image]

Maybe it has to do to the fact I have found and installe Muse Direct too...

closed 2022-03-23 by Zibri

jdpigeon, 2022-03-23:
It looks like there was some issue with the way the LSL library interacted with your network environment.

I'm glad to hear things seem to be working now!