alexandrebarachant/muse-lsl #179: Issues with muses

itsoulos, 2022-03-29:
Dear all,
I have the following errors on Debian Linux with Muse S
What is wrong with my device?

 muselsl stream --name Muse-4899
Backend was 'auto' and bluetoothctl was found, using to list muses...
Searching for Muses, this may take up to 10 seconds...
Found device Muse-4899, MAC Address 00:55:DA:B9:48:99
Connecting to Muse-4899: 00:55:DA:B9:48:99...
Traceback (most recent call last):
  File "/usr/local/bin/muselsl", line 33, in <module>
    sys.exit(load_entry_point('muselsl==2.2.0', 'console_scripts', 'muselsl')())
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/__main__.py", line 57, in main
    cli = CLI(args.command)
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/cli.py", line 9, in __init__
    getattr(self, command)()
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/cli.py", line 106, in stream
    stream(args.address, args.backend, args.interface, args.name, args.ppg,
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/stream.py", line 221, in stream
    didConnect = muse.connect()
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/muse.py", line 90, in connect
    self._subscribe_eeg()
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/muse.py", line 279, in _subscribe_eeg
    self.device.subscribe(MUSE_GATT_ATTR_TP9, callback=self._handle_eeg)
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/backends.py", line 66, in subscribe
    _wait(self._client.start_notify(uuid, wrap))
  File "/usr/local/lib/python3.9/dist-packages/muselsl-2.2.0-py3.9.egg/muselsl/backends.py", line 11, in _wait
    return loop.run_until_complete(coroutine)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.9/dist-packages/bleak-0.14.2-py3.9.egg/bleak/backends/bluezdbus/client.py", line 915, in start_notify
    raise BleakError(
bleak.exc.BleakError: Characteristic with UUID 273e0003-4c4d-454d-96be-f03bac821358 could not be found!

jdpigeon, 2022-03-29:
What happens if you try annother backend `--backend gatt`?

itsoulos, 2022-03-30:
Good morning. Finally the problem has resolved after updating some debian packages and using the gatt interface.

xloem, 2022-12-11:
I am not sure yet what causes this, but I am experiencing it resolving after retrying a few times.