alexandrebarachant/muse-lsl #185: How can I get data with (muselsl view --version 2) my muse 2?

jongnmi, 2022-09-22:
Thanks for your coding. I successly connected my muse 2 and my notebook. So I can see the result window including TP9, AF7, AF8, TP10 graph. But I want to use this graph in real time. How Can I get the data the number of Hz?(on the right side in the window)
Its really appreciate if you tell me the way how to solve this problem.

In summary, I can see the graph moving window in real time, but I want to get data that I can use in my research.

jdpigeon, 2022-09-22:
What muselsl does is open up a Lab Streaming Layer Stream that emits EEG data from the Muse headband. If you're interested in writing your own code to analyze data in real time, you will simply need to use pylsl to connect to the EEG stream that the `muselsl stream` command creates.

Something like this example should get you started: https://github.com/labstreaminglayer/liblsl-Python/blob/master/pylsl/examples/ReceiveData.py

Colton127, 2022-09-23:
You can use neurofeedbacklab to get the spectral power of set frequencies and channels in real-time: https://github.com/arnodelorme/neurofeedbacklab

closed 2022-09-28 by jdpigeon