alexandrebarachant/muse-lsl #189: TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicBlueZDBus' and 'int'

GabrieleFerrero, 2022-10-23:
Hi, I tried to install muselsl library but when I try to run a stream it gives me this error:
```
Connecting to Muse: 00:55:DA:B9:67:2E...
Connected.
Streaming EEG...
ERROR:root:A message handler raised an exception: unsupported operand type(s) for +: 'BleakGATTCharacteristicBlueZDBus' and 'int'.
Traceback (most recent call last):
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/dbus_fast/message_bus.py", line 778, in _process_message
    result = user_handler(msg)
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/bleak/backends/bluezdbus/manager.py", line 832, in _parse_msg
    on_value_changed(message.path, self_interface["Value"])
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/bleak/backends/bluezdbus/client.py", line 152, in on_value_changed
    callback(bytearray(value))
  File "/home/gabriele/Applicazioni/anaconda3/envs/prova/lib/python3.7/site-packages/muselsl/backends.py", line 63, in wrap
    value_handle = declaration_handle + 1
TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicBlueZDBus' and 'int'
```

jdpigeon, 2022-10-25:
My guess is that this will be solved by this PR: https://github.com/alexandrebarachant/muse-lsl/pull/187

davidbichara, 2022-10-26:
I am having the same issue listed in this issue. @jdpigeon is any additional support needed to get the PR pass builds? I only ask because I need the fix for a time sensitive project and would love to help contribute if there is something I can do to make the release easier. Let me know!

xloem, 2022-12-11:
If the pr is in question, a version specifier of '`bleak<0.18.0` might be appropriate in setup.py .

fullflyt, 2022-12-27:
I've tried a lot of python code including the code found here, like neurofeedback.py, startMuseStream.py, recordStream.py and muselsl view. I'm using a ubuntu installation. I've also tried some on a Mac M1 laptop. Some successes, but mostly failures. I also switch bluetooth on and off. Bleak is set to 0.17.0. 
- pylsl does not seem to work.
- muselsl and muselsl view seem to work for startMuseStream.py, not recordStream.py
- utils.py is installed
- neurofeedback.py does not work. it can't find EEG stream.
- Typical errors are "IndexError: list index out of range" or can't find EEG stream. The Bleak and int errors are gone.

Any ideas for whats wrong here. Some other packages that have a version number that does not work?
[piplist.txt](https://github.com/alexandrebarachant/muse-lsl/files/10311104/piplist.txt)

xloem, 2022-12-28:
@fullflyt I recommend opening a new issue and including very clear logs of output and commands run, as well as links to anything involved outside this repository.

It can take a long time to get a response in this repo.

jdpigeon, 2023-01-03:
https://github.com/alexandrebarachant/muse-lsl/pull/187 has been merged and the package has been pushed to pypi. Thanks for all the attention here everybody.

closed 2023-01-03 by jdpigeon

fullflyt, 2023-01-05:
Hi Alexandre,

When I ran this first:
$ python startMuseStream.py
Backend was 'auto' and bluetoothctl was found, using to list muses...
Searching for Muses, this may take up to 10 seconds...
Found device Muse-37A5, MAC Address 00:55:DA:B7:37:A5
Connecting to Muse: 00:55:DA:B7:37:A5...
Connected.
Streaming EEG...
^CDisconnected.
Stream has ended

all other commands work:
muselsl view
python neurofeedback.py
pylsl

Thus I don't need to report anything.

Next up is to calculate coherence between the two front sensors and plot
streams of front sensors and coherence, like this:
https://www.youtube.com/watch?v=XNkir2vzq3o

I'll also test the latest version of Bleak.
$ pip show Bleak
Name: bleak
Version: 0.19.5
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: ***@***.***
License: MIT
Location: /home/einar/.local/lib/python3.10/site-packages
Requires: async-timeout, dbus-fast
Required-by: muselsl

It didn't work, so I'll return to 0.17.0.

Med vennlig hilsen
Einar Jørgensen
Daglig Leder
Fullflyt AS

Mobil: +47 982 33 894 <982%2033%20894>
email: einar@ ***@***.***>fullflyt.no