alexandrebarachant/muse-lsl #191: Find yes, stream no

rpgrunwald, 2022-11-21:
`(base) C:\Users\Paul>muselsl list
Searching for Muses, this may take up to 10 seconds...
Found device Muse-3305, MAC Address 00:55:DA:B5:33:05

(base) C:\Users\Paul>muselsl stream
Searching for Muses, this may take up to 10 seconds...
No Muses found.

(base) C:\Users\Paul>muselsl stream --name Muse-3305
Searching for Muses, this may take up to 10 seconds...
No Muses found.`

xloem, 2022-12-11:
Hi, I developed the Bleak backend your system is using, and I have not yet ever tested and troubleshooted it on Windows. I am so sorry about this. You could try a different backend.

I do not have a Windows machine set up at the moment, but please still ping me when errors come up inside it, to help me remember the importance of getting one.

jack-moore-psymplicity, 2023-01-03:
Same error different issue. Any help would be much appreciated.

`Searching for Muses, this may take up to 10 seconds...
Found device Muse-BAF7, MAC Address 00:55:DA:B7:BA:F7
Connecting to Muse: 00:55:DA:B7:BA:F7...
2023-01-03 14:44:07.260 (  11.883s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{537B6D84-638E-4E68-8BC0-21DCBB7EC893}' (status: 2, multicast: 1
2023-01-03 14:44:07.260 (  11.884s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{80512BBD-B14C-48B9-AD24-6B28881005CB}' (status: 2, multicast: 1
2023-01-03 14:44:07.260 (  11.884s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{345AA57B-2295-4E61-B8F3-63A8E6D0DFED}' (status: 2, multicast: 1
2023-01-03 14:44:07.260 (  11.884s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{C09905C2-5E85-419B-A475-1522506C8375}' (status: 1, multicast: 1
2023-01-03 14:44:07.260 (  11.884s) [        7892C2E5]      netinterfaces.cpp:58    INFO| 	IPv6 ifindex 14
2023-01-03 14:44:07.260 (  11.884s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{A0D6D37E-3BD4-41F2-8544-1BEE03AEA19A}' (status: 2, multicast: 1
2023-01-03 14:44:07.261 (  11.884s) [        7892C2E5]      netinterfaces.cpp:36    INFO| netif '{BB1D44F7-778B-11EC-A99C-806E6F6E6963}' (status: 1, multicast: 1
2023-01-03 14:44:07.261 (  11.885s) [        7892C2E5]      netinterfaces.cpp:58    INFO| 	IPv6 ifindex 1
2023-01-03 14:44:07.261 (  11.885s) [        7892C2E5]         api_config.cpp:270   INFO| Loaded default config
2023-01-03 14:44:07.261 (  11.885s) [        7892C2E5]             common.cpp:65    INFO| git:5eded5c1d381a1a5fbbcce105edfaa53f009176a/branch:refs/tags/v1.16.0/build:Release/compiler:MSVC-19.0.24245.0/link:SHARED
2023-01-03 14:44:07.262 (  11.886s) [        7892C2E5]         udp_server.cpp:82    WARN| Could not bind multicast responder for ff02:113d:6fdd:2c17:a643:ffe2:1bd1:3cd2 to interface ::1 (An invalid argument was supplied.)
2023-01-03 14:44:07.262 (  11.886s) [        7892C2E5]         udp_server.cpp:82    WARN| Could not bind multicast responder for ff05:113d:6fdd:2c17:a643:ffe2:1bd1:3cd2 to interface ::1 (An invalid argument was supplied.)
Connected.
Streaming EEG...
Exception in callback BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2D88>)(bytearray(b'\...\xb0\xb0\x91')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62
handle: <Handle BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2D88>)(bytearray(b'\...\xb0\xb0\x91')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62>
Traceback (most recent call last):
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py", line 63, in wrap
    value_handle = declaration_handle + 1
TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicWinRT' and 'int'
Exception in callback BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2D08>)(bytearray(b'\...afK\xe1|\x85')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62
handle: <Handle BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2D08>)(bytearray(b'\...afK\xe1|\x85')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62>
Traceback (most recent call last):
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py", line 63, in wrap
    value_handle = declaration_handle + 1
TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicWinRT' and 'int'
Exception in callback BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2BC8>)(bytearray(b'\...\x00\x08\xd9')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62
handle: <Handle BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2BC8>)(bytearray(b'\...\x00\x08\xd9')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62>
Traceback (most recent call last):
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py", line 63, in wrap
    value_handle = declaration_handle + 1
TypeError: unsupported operand type(s) for +: 'BleakGATTCharacteristicWinRT' and 'int'
Exception in callback BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2C88>)(bytearray(b'\...x00\x00\x07A')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62
handle: <Handle BleakDevice.subscribe.<locals>.wrap(<bleak.backen...001EE68FA2C88>)(bytearray(b'\...x00\x00\x07A')) at C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py:62>
Traceback (most recent call last):
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\asyncio\events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "C:\Users\JackMoore\AppData\Local\Programs\Python\Python37\lib\site-packages\muselsl\backends.py", line 63, in wrap
    value_handle = declaration_handle + 1`

jdpigeon, 2023-01-03:
> Same error different issue. Any help would be much appreciated.

This error is different and should be resolved by https://github.com/alexandrebarachant/muse-lsl/pull/187

As for the original issue, this MAC address looks suspicious to me

> Found device Muse-nnnn, MAC Address 00:uu:xxx:yy:zz:00

Is there something peculiar going on with this machine? Perhaps it's running via virtual machine?

jack-moore-psymplicity, 2023-01-04:
Hi @jdpigeon  thanks for getting back to me. I solved the issue!!
I changed the version of bleak I had from 18.0.0 to 17.0....that has fixed the problem