alexandrebarachant/muse-lsl #201: muselsl view command not working.

jessiusmitchellius, 2023-09-23:
[image]

aureateAnatidae, 2023-09-23:
just to make sure, other commands work? since the program says "muselsl not found", i might suspect you havent run `pip install muselsl`, or that something has gone wrong during installation.

please try `muselsl list' and `pip show muselsl`, then show us what it returns.

aureateAnatidae, 2023-09-23:
actually, could you show us what `pip -V ` returns as well?

jessiusmitchellius, 2023-09-23:
I have run pip install muselsl 

muselsl list shows the same thing

this is pip-v
[image]

aureateAnatidae, 2023-09-23:
thank you for the fast reply. **since you're on windows, you might consider BlueMuse, it seems like a far less frustrating alternative to getting bluetooth working in a windows terminal.**

i've gone on my windows machine and seen that it calls fine for pip 23.2.1 and python 3.8.10 in a bash shell (later returning a big bluetooth error about missing site packages). i've also tried similar to your setup with pip 23 and python 3.11.4 in command line, then i get the same error as yours. i want to hypothesize that it's looking somewhere it isn't installed.

i'm not good enough at this to get you a proper solution, but in the meantime, check out BlueMuse or some other lsl streaming software. i've gone ahead and confirmed that BlueMuse still installs properly and works.