alexandrebarachant/muse-lsl #208: Can't see my muse in the list

seenaimul, 2024-02-18:
Hi all,
This is the first time I'm using Muse2 and need your help.
I am using Windows 11 and not using any additional Bluetooth dongle.
As suggested, I have set up BlueMuse and can see my Muse2 connected and can do streaming.
[image]

Now when I'm trying to list the available muses using the '$ muselsl list' command it can't detect any muse. I also changed the default 10s timeout to 60s but I still can't detect it.
[image]

I have installed 'pip install muselsl' the required libraries and the following is the list of libraries in my virtual environment of the project.
$ pip list
Package                                                 Version
------------------------------------------------------- -----------
bitarray                                                2.9.2
bitstring                                               4.1.4
bleak                                                   0.21.1
contourpy                                               1.2.0
cycler                                                  0.12.1
enum-compat                                             0.0.3
fonttools                                               4.49.0
joblib                                                  1.3.2
kiwisolver                                              1.4.5
matplotlib                                              3.8.3
muselsl                                                 2.2.2
numpy                                                   1.26.4
packaging                                               23.2
pandas                                                  2.2.0
pexpect                                                 4.9.0
pillow                                                  10.2.0
pip                                                     24.0
ptyprocess                                              0.7.0
pygatt                                                  4.0.5
pylsl                                                   1.16.2
pyparsing                                               3.1.1
pyserial                                                3.5
python-dateutil                                         2.8.2
pytz                                                    2024.1
scikit-learn                                            1.4.1.post1
scipy                                                   1.12.0
seaborn                                                 0.13.2
six                                                     1.16.0
threadpoolctl                                           3.3.0
tzdata                                                  2024.1
winrt-runtime                                           2.0.0b1
winrt-Windows.Devices.Bluetooth                         2.0.0b1
winrt-Windows.Devices.Bluetooth.Advertisement           2.0.0b1
winrt-Windows.Devices.Bluetooth.GenericAttributeProfile 2.0.0b1
winrt-Windows.Devices.Enumeration                       2.0.0b1
winrt-Windows.Foundation                                2.0.0b1
winrt-Windows.Foundation.Collections                    2.0.0b1
winrt-Windows.Storage.Streams                           2.0.0b1
(venv)

Can you please guide me on how I can **resolve this issue and see the live data visualisation while using the Muse** like the sample image shown in the repository?
[image]

chadcwilliams, 2024-03-01:
This occurs because your BlueMuse is open, which for some reason makes it undetectable by muselsl's list function. When you open BlueMuse, you'll notice that the lights on your Muse go solid (I have a Muse S, so I am not 100% sure that's true for the Muse 2), which means that BlueMuse has connected to it, making it unavailable for muselsl to find.

The work-around is to simply have BlueMuse closed when running the muselsl functions, as the muselsl stream function `muselsl stream --backend bluemuse` will open BlueMuse and then connect to the device automatically for you.

Neuroera, 2025-02-27:
Is bled112 needed to connect Muse 2016 with Windows 10? where can I get it? thanks