alexandrebarachant/muse-lsl #216: syntax error

danielemarinazzo, 2024-10-10:
Mac M2 OS 14.3.1 (23D60)

when running "muselsl" command we get a syntax error message (corresponding to the colon after "device")

  File "/Users/USERNAME/miniconda3/envs/eeg-notebooks/lib/python3.7/site-packages/muselsl/muse.py", line 90
    if ((device := self.adapter.connect(self.address, retries))

ErikBjare, 2024-10-18:
Requires Python 3.8+: https://github.com/NeuroTechX/EEG-ExPy/issues/277#issuecomment-2422786165

ErikBjare, 2024-10-18:
I think the syntax should be changed to support Python 3.7 since that's trivial.

Not up-to-date with things in EEG-ExPy, but I think it might be pretty tied to Python 3.7 still due all the shenanigans in PsychoPy: https://www.psychopy.org/download.html#linux

danielemarinazzo, 2024-10-18:
Ok, thanks. So there's a solution to use the package as it is? If necessary bypassing muselsl and using brainflow only (even iv on some laptops only the former works for some reason)?

Neuroera, 2025-02-27:
Is bled112 needed to connect Muse 2016 with Windows 10? where can I get it? thanks