alexandrebarachant/muse-lsl #222: Muse S Athena Support

BrianMohseni, 2025-06-10:
Muse S Athena has issues connecting with Muse LSL, even for pre-existing channels, due to some changes made (I believe in naming convention). Also added FNIRS.

Not in any rush, but I was wondering if support for Muse S Athena was being worked on.

Thanks for your hard work thus far!

tjthejuggler, 2025-07-22:
Is there any update on this? I want to get one, but only once I can do stuff with it.

kowalej, 2025-08-10:
Anyone have more info on the Athena? I have at least one BlueMuse user asking to support the device, but I don't have access to the hardware. I've tried to get some more info, here's the thread: https://github.com/kowalej/BlueMuse/issues/73. I could probably help make adjustments if needed to Muse-LSL as well.

Here's the screenshot the user sent with the GATT characteristics, I'm seeing the same data service as always, but with way less than expected channels showing up.

<img width="3840" height="2032" alt="Image" src="https://github.com/user-attachments/assets/810560d6-85da-496e-9dbe-466e455ec847" /

BrianMohseni, 2025-08-25:
> Anyone have more info on the Athena? I have at least one BlueMuse user asking to support the device, but I don't have access to the hardware. I've tried to get some more info, here's the thread: [...]

update: Muse S Athena is now supported by the 'brainflow' module. it has not yet received support in MuseLSL.

thenoobmlengineer, 2025-08-25:
Im trying to connect muse S athena using brainflow now, and Im getting errors.